stereolabs/zed-matlab #1: `GLIBCXX_3.4.19' not found

aseyfi, 2016-07-11:
Hi,

I am using Matlab R2016a on Ubuntu 14.04, with gcc and g++ 4.7.4. When I run ZED_Camera.m, I get this error:

>
=========SL_ZED_WITH_MATLAB=========
Invalid MEX-file '/home/ahmad/work/zed-matlab/matlab/mexZED.mexa64': /usr/local/MATLAB/R2016a/bin/glnxa64/../../sys/os/glnxa64/libstdc++.so.6: version
`GLIBCXX_3.4.19' not found (required by /usr/local/zed/lib/libsl_tracking.so)

Error in ZED_Camera (line 11)
mexZED('create', 720, 60);
> 

Any suggestions?
Thank You,

P-yver, 2016-07-12:
hi @aseyfi,

It seems that your version of gcc is old, have you try to update it ?
Or may be this [link](http://askubuntu.com/a/582910) could help you.

revilokeb, 2016-08-02:
@aseyfi those suggestions helped me: https://github.com/mlapin/libsdca/issues/1. More specifically I could run ZEDcamera.m after starting Matlab as follows: `LD_PRELOAD=/PATH/TO/libstdc++.so.6 matlab&`, with /PATH/TO found by `locate libstdc++.so.6` (in my case /usr/lib/x86_64-linux-gnu/)

commit 16e8a5cf (2016-10-14): Fix #1 instructions for LD_PRELOAD
closed 2016-10-14 by adujardin