stereolabs/zed-matlab #5: obtain depth map from only left and right color images

nikky4D, 2017-12-11:
Hi,

I have the left and right images from the zed explorer. I would like to get the depth images of these scenes in MATLAB. Unfortunately, I don't have access to the environment where the images were taken. Is there a way to get the depth images when I only have the left and right images from the zed? 

I have the camera parameters fx, fy, cx,cy, baseline. 

Right now, I am assuming these images are already rectified. So, I can get the disparity map, and calculate the depth like so: 

disparityMap = disparity(img_left, img_right);
figure;imshow(disparityMap,[0, 100],'InitialMagnification', 50);

%% By stereoLabs  calculation, Z is:
fx_left = 699.291; baseline = 119.923;
Z = (fx_left .* baseline ) ./ (1.*disparityMap);
figure;imshow(Z,[0, 5000]);

is this the correct method? 

Thanks

obraun-sl, 2017-12-12:
Hi,
your formulae is correct (f*B/disp) but ZED Explorer gives unrectified images, therefore you won't have good results.

2 solutions : 

* Use the parameters from the calibration file (fx/y, cx/y, rx,cv,rz) and create a rectification map and a new camera matrix. You can then ingest the rectified images in disparity() function from matlab. I don't know how to do it in Matlab.
* Use the ZED SDK to get rectified images (retrieveImage()) and new calibration parameters (getCameraInformation().calibration_parameters) you can directly ingest in disparity()

/*OB*/

ASingh245, 2021-10-04:
Hi,
I am using ZED2 camera and using the APIs as mentioned below to fetch the depth. And then using the point cloud to calculate depth.
1. zed.retrieveImage(image, VIEW::LEFT);
2. zed.retrieveMeasure(depth, MEASURE::DEPTH); to fetch depthmap.  
3. zed.retrieveMeasure(point_cloud, MEASURE::XYZRGBA);
float distance = sqrt(point3D.x*point3D.x + point3D.y*point3D.y + point3D.z*point3D.z);

I have few questions as:
1. Is there any reason to not use these APIs and calculating with disparity?
2. Why we are using only left view? Should I use both right and left view?