stereolabs/zed-matlab #8: WARNING: FPS is too low to enable positional tracking and spatial mapping. Consider using PERFORMANCE parameters

sophie-shin, 2018-08-17:
I got this messages from the command window.

```
=========SL_ZED_WITH_MATLAB -- Point Cloud=========
ZED (Init) >> Depth mode: PERFORMANCE
ZED (Init) >> Video mode: HD720@60

result =

    'SUCCESS'

image_size =

        1280
         720

ZED (Init) >> WARNING: FPS is too low to enable positional tracking and spatial mapping. Consider using PERFORMANCE parameters
ZED (Init) >> WARNING: FPS is too low to enable positional tracking and spatial mapping. Consider using PERFORMANCE parameters

```

Any codes were not changed from the initial code.
Zedfu works well.

Please help!

willshw, 2018-11-09:
I'm experiencing this too. My ZED camera is connected to Jetson TX2. Even without rviz open, rostopic hz will show the /point_cloud/cloud_registered was publishing about 10Hz. Did you find the solution?

willshw, 2018-11-10:
nvm, I think the new package update fixed the issue.

> I'm experiencing this too. My ZED camera is connected to Jetson TX2. Even without rviz open, rostopic hz will show the /point_cloud/cloud_registered was publishing about 10Hz. Did you find the solution?

closed 2022-04-26 by github-actions[bot]