stereolabs/zed-matlab #11: CMake error while configuration

FabianAC07, 2019-06-13:
So, I am trying to build the CMake file but I get the following error:

`CMake Error at CMakeLists.txt:5 (message):
  You've selected the 32bit version of Visual Studio 15 2017.

   Please delete the cache (file->Delete Cache) and use the 64bit version. (Visual Studio 15 2017 Win64)`

Even though, I select x64 over the platform generator.

[image]

The whole command window as follow:

`Selecting Windows SDK version 10.0.14393.0 to target Windows 10.0.17763.
The C compiler identification is MSVC 19.10.25027.0
The CXX compiler identification is MSVC 19.10.25027.0
Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.10.25017/bin/HostX86/x64/cl.exe
Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.10.25017/bin/HostX86/x64/cl.exe -- works
Detecting C compiler ABI info
Detecting C compiler ABI info - done
Detecting C compile features
Detecting C compile features - done
Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.10.25017/bin/HostX86/x64/cl.exe
Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.10.25017/bin/HostX86/x64/cl.exe -- works
Detecting CXX compiler ABI info
Detecting CXX compiler ABI info - done
Detecting CXX compile features
Detecting CXX compile features - done
CMake Error at CMakeLists.txt:5 (message):
  You've selected the 32bit version of Visual Studio 15 2017.

   Please delete the cache (file->Delete Cache) and use the 64bit version. (Visual Studio 15 2017 Win64)

Configuring incomplete, errors occurred!
See also "C:/Users/fabia/Desktop/zed_matlab/zed-matlab/build/CMakeFiles/CMakeOutput.log".`

Can someone help me to figure out what is wrong?

Kurkyxp, 2019-06-15:
Hey. I have some issues to set this with VS2019 and MATLAB2018b.

However i figured out this step solution.

FROM YOUR PATH:
C:/Users/fabia/Desktop/zed_matlab/zed-matlab/src/CMakeLists.txt

REMOVE the condition:
IF(WIN32 AND NOT CMAKE_GENERATOR MATCHES "[a-z A-Z 0-9]*64" )
    message(FATAL_ERROR "You've selected the 32bit version of ${CMAKE_GENERATOR}. \n Please delete the cache (file->Delete Cache) and use the 64bit version. (${CMAKE_GENERATOR} Win64)")
ENDIF()

SAVE the file and RECOMPILE. Some documentation out there says that now VS select x64 or x86 based on your system. I guess you're x64 so the condition would be unnecessary.

P-yver, 2019-06-19:
Hi @FabianAC07 ,
I have updated the cmake file to resolve this issue, can you try it and confirm?

FabianAC07, 2019-06-20:
Hello @P-yver, 

Thanks for updating the cmake file, but now there is another issue, the cmake file is not finding the MATLAB_ROOT env variable... even though I have added it... 

```
Selecting Windows SDK version 10.0.14393.0 to target Windows 10.0.17763.
The C compiler identification is MSVC 19.10.25027.0
The CXX compiler identification is MSVC 19.10.25027.0
Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.10.25017/bin/HostX86/x64/cl.exe
Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.10.25017/bin/HostX86/x64/cl.exe -- works
Detecting C compiler ABI info
Detecting C compiler ABI info - done
Detecting C compile features
Detecting C compile features - done
Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.10.25017/bin/HostX86/x64/cl.exe
Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.10.25017/bin/HostX86/x64/cl.exe -- works
Detecting CXX compiler ABI info
Detecting CXX compiler ABI info - done
Detecting CXX compile features
Detecting CXX compile features - done
MATLAB_ROOT environment variable not set.
In Linux this can be done in your user .bashrc file by appending the corresponding line, e.g:
export MATLAB_ROOT=/usr/local/MATLAB/R2014b
In Windows this can be done by adding system variable, e.g:
MATLAB_ROOT=C:/Program Files/MATLAB/R2014b
CMake Error at CMakeLists.txt:30 (message):
  MATLAB not found...nothing will be built.

Configuring incomplete, errors occurred!
See also "C:/Users/fabia/Desktop/zed_matlab/zed-matlab/src/build/CMakeFiles/CMakeOutput.log".
```
Here the screenshot from the environment variables... I even added it to the main path...

[image]

Any suggestion?

P-yver, 2019-06-21:
The var __MATLAB_ROOT__ is not an environment variable, it is a CMAKE variable.
Have you look to [our documentation](https://www.stereolabs.com/docs/matlab/#using-the-zed-sdk), you can set  __MATLAB_ROOT__ directly with Cmake UI.

FabianAC07, 2019-06-21:
Hello @P-yver 

The variables in documentation do not appear on the CMake GUI, so I do not know how to include the matlab path... 

[image]

Same error as my previous post...

P-yver, 2019-06-24:
check the **Grouped** and **Advanced** checkboxes  on the cmake panel, you will see under the title __Ungrouped Entries__ the cmake variable **MATLAB_ROOT** where you can set your matlab path.

[image]

FabianAC07, 2019-06-24:
Hello @P-yver, 

Thank you very much for the support. It finally worked!

Regards!

closed 2019-06-24 by FabianAC07