stereolabs/zed-matlab #16: how to compatibility with matlab 2019b on linux

hzm2016, 2020-01-31:


P-yver, 2020-02-11:
Hi,
Have you tried our build [instructions](https://github.com/stereolabs/zed-matlab#build-the-program)?

fabianmpa, 2020-02-23:
Hi, i am having the same problem. I've already tried the build instructions. Mine is a Zed Mini and i have installed SDK 3.0.

P-yver, 2020-02-25:
Hi,
can you share the errors you have?
Does came from the CMake or from the project build?

fabianmpa, 2020-02-27:
They come when i try to run .m scripts on matlab 
For example, if i run the ZED_Camera.m i get this error:

========= ZED SDK PLUGIN =========
-- Retrieve Images and Depth --
Invalid MEX-file '/home/fabian/zed/zed-matlab-master/matlab/mexZED.mexa64':
/home/fabian/zed/zed-matlab-master/matlab/mexZED.mexa64: undefined symbol:
_ZN2cv3Mat20updateContinuityFlagEv

Error in ZED_Camera (line 17)
result = mexZED('open', InitParameters);

fabianmpa, 2020-03-27:
Update: Works fine wih SDK 3.1

closed 2020-11-13 by P-yver