stereolabs/zed-matlab #21: build 'mexZED' and 'INSTALL' have mistakes using the new updata zed-matlab

ultimata, 2020-11-21:
I install 'ZED_SDK_Windows10_cuda11.1_v3.3.1' and 'cuda_11.1.0_456.43_win10' and 'vs2015'.
The "ZED Depth Viewer" can work.
I open 'MEX.sln' using vs2015, and choose 'release' and 'x64'.
When I build 'mexZED', there are some mistakes. The mistakes are shown below:
1.
[image]
"C2440	“static_cast”: can not convert from “double”to“sl::REFERENCE_FRAME”	mexZED	F:\ZED_MATLAB\zed-matlab-master20201116\src\mex\mexZED.cpp	723

in 'mexZED.cpp' and line 723, the program is 'ref_f =  static_cast<sl::REFERENCE_FRAME>(mxGetPr(prhs[1])[0]);'
[image]

2.when I build ' INSTALL ', there are mistakes.

MSB3073	命令“setlocal
D:\CMake\bin\cmake.exe -DBUILD_TYPE=Release -P cmake_install.cmake
if %errorlevel% neq 0 goto :cmEnd
:cmEnd
endlocal & call :cmErrorLevel %errorlevel% & goto :cmDone
:cmErrorLevel
exit /b %1
:cmDone
if %errorlevel% neq 0 goto :VCEnd
:VCEnd”已退出，代码为 1。	INSTALL	C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.CppCommon.targets	133
[image]

I need help to solve the mistakes. 
Thanks.

P-yver, 2020-11-23:
Can you try first to cast values to int and then to the required type?

```cpp
if (nrhs == 2) {
    int val = mxGetPr(prhs[1])[0];
    ref_f = static_cast<sl::REFERENCE_FRAME>(val);
}
```
and

```cpp
if (nrhs == 2) {
    int val = mxGetPr(prhs[1])[0];
    t_ref = static_cast<sl::TIME_REFERENCE>(val);                
}
```

ultimata, 2020-11-24:
Hi, P-yver
thanks your help. I rewrite the code as your provied.
and build the mexZED. there are some mistakes such as sl::depth_mode in line 343 (C2440)of mexZED.cpp.
I add code as your advised.
```
if(ref == curr) {
        if(mxIsNumeric(v_in)){
           int val = mxGetPr(v_in)[0];
            output = static_cast<T>(**val**);
            //mexPrintf("Set %s\n", curr.c_str());

        }
        else
            mexPrintf("Can not convert %s\n", curr.c_str());
   }
```
and I rebuild the mexZED, it can work.
there are other error in ZED_Camera.m, when set 'InitParameters.camera_resolution = 2; %HD720',There is a warning in matlab:
========= ZED SDK PLUGIN =========
-- Retrieve Images and Depth --
warning: ZED SDK error when calling: open, Error: CAMERA FAILED TO SETUP 
========= END =========
**And  I set ‘InitParameters.camera_resolution = 3’ ，it can work.**

The value of InitParameters.camera_resolution is set to 0,1,2, the Error of matlab is ‘CAMERA FAILED TO SETUP’

Can you give some advice?

Thanks.

P-yver, 2020-11-24:
Hi,
Can you run the program `ZED Explorer` provided in the installer and check if you can switch between resolutions without issues or run the `ZED Diagnostic` tool?

ultimata, 2020-11-28:
Hi, P-yver
Thanks your help.
I connect the camera with usb 2.0, so the resolution do not work.
I use usb 3.0 to connect the camera and the camera can work.
Thanks again.

closed 2022-04-26 by github-actions[bot]