stereolabs/zed-matlab #25: Getting errors building

RickFSS, 2023-04-30:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

Getting Compile errors on Windows 11 with Matlab 2023a VS 2022

Severity	Code	Description	Project	File	Line	Suppression State
Error (active)	E0135	class "sl::ObjectData" has no member "keypoint_2d"	mexZED	D:\Projects\zed-matlab\src\mex\mexZED.cpp	211	

### Steps to Reproduce

1. Install SDK
2. clone matlab ZED repo
3. Run CMake pointing to matlab install
4. Configure (ok)
5. Generate (ok)
6. Try to build - Get error
7.
8.
...

### Expected Result

Build without errors.

### Actual Result

Multiple compile errors.

### ZED Camera model

ZED2

### Environment

```shell
OS Windows
```

P-yver, 2023-05-03:
Hi,
I just update the code base to support the ZED SDK 4.0.
Where Object Detection and Body Tracking are now in separated functions.

https://www.stereolabs.com/developers/release/

Ibrahimsh104, 2024-07-25:
I have issue with cmake i gives me error of cuda :
[image]

i reinstalled cuda "CUDA Toolkit 12.5" multiple times and i notesed it gives me this message : 
Installed:
     - Nsight for Visual Studio 2022
     - Nsight for Visual Studio 2019
     - Nsight Monitor
Not Installed:
     - Nsight for Visual Studio 2017
       Reason: VS2017 was not found
     - Integrated Graphics Frame Debugger and Profiler
       Reason: see https://developer.nvidia.co
     - Integrated CUDA Profilers
       Reason: see https://developer.nvidia.com/nsight-vstools
[image]

i tried to solve it and installed nsight 
NVIDIA Nsight Visual Studio Edition:https://developer.nvidia.com/nsight-visual-studio-edition
NVIDIA Nsight Systems:https://developer.nvidia.com/nsight-systems
NVIDIA Nsight Compute:https://developer.nvidia.com/nsight-compute
NVIDIA Nsight Graphics:https://developer.nvidia.com/nsight-graphics
and video-codec-sdk "https://developer.nvidia.com/video-codec-sdk" bec. i noticed it is messing in the cmake environment and also not found at the location of cuda 
how can i solve this problem?