stereolabs/zed-matlab #26: errors when make the files

Nikolatesla-lj, 2023-06-11:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

I followed the build procedure provided on your zed website: export MATLAB_ROOT=/usr/local/MATLAB/R2016b
mkdir build
cd build
cmake ... /src
make
make install
However, when the "make" step is executed, the following error occurs:

### Steps to Reproduce

I followed the build procedure provided on your zed website: export MATLAB_ROOT=/usr/local/MATLAB/R2016b
mkdir build
cd build
cmake ... /src
make
make install
However, when the "make" step is executed, the following error occurs:

### Expected Result

Compilation passes normally

### Actual Result

/home/ll/Documents/MATLAB/zed-matlab/src/mex/mexZED.cpp: In function ‘void mexFunction(int, mxArray**, int, const mxArray**)’:
/home/ll/Documents/MATLAB/zed-matlab/src/mex/mexZED.cpp:726:98: error: taking address of rvalue [-fpermissive]
  726 |  &(calibration_parameters.stereo_transform.getRotationVector()), sizeof(sl::float3));
      |   ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~

/home/ll/Documents/MATLAB/zed-matlab/src/mex/mexZED.cpp:730:98: error: taking address of rvalue [-fpermissive]
  730 | n), &(calibration_parameters.stereo_transform.getTranslation()), sizeof(sl::float3));
      |      ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~

make[2]: *** [mex/CMakeFiles/mexZED.dir/build.make:76: mex/CMakeFiles/mexZED.dir/mexZED.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:98: mex/CMakeFiles/mexZED.dir/all] Error 2
make: *** [Makefile:136: all] Error 2

### ZED Camera model

ZED2i

### Environment

```shell
OS:ubuntu 22
zed sdk: 4.03
opencv: 4.6
cuda:12.1
```

Nikolatesla-lj, 2023-06-11:
by the way the zed-matlab version is the latest version

P-yver, 2023-06-27:
Hi,
indeed there is an issue for linux.
It should be fixed with this commit https://github.com/stereolabs/zed-matlab/commit/f573f10be3724d38a622cfb3629a6247deed4085 (just pull the current branch).

closed 2023-07-01 by Nikolatesla-lj